SkyNet3D/anet-board #9: Make compatible with Arduino IDE Boards Manager?

benlye, 2017-07-05:
What do you think about making the Github repo compatible with the Arduino IDE boards manager?

I had a go at it here:
https://github.com/benlye/anet-board/tree/board-manager

It requires changing the file layout and creating the JSON index file.  The index file I created works, but is a mix of information - it points to my repo only because of the file structure changes that needed to be in the download archive, and because I had to tag a release.

It would mean that any future changes to the board definition would cause people to get a pop-up in the IDE telling them there was an update available.

Ben

benlye, 2017-07-05:
You can try it out by adding this URL as an 'Additional Board Manager URL' in the IDE preferences:
https://raw.githubusercontent.com/benlye/anet-board/board-manager/package_skynet3d_anet_board_index.json

oderwat, 2017-07-05:
Hi Ben. Yeah thats cool. I am still on vacation and therefor limited with testing stuff. I will come back to it in about a week!

Renaud11232, 2019-03-28:
Any news about this ?
Being able to install with the board manager would be very nice :)

benlye, 2019-03-29:
I don't know if @oderwat is maintaining this any more.  I've tweaked my fork to provide a Boards Manager compatible verison.  Instructions to use it are included in the readme - take a look: https://github.com/benlye/anet-board

oderwat, 2019-03-29:
Nope ... after switching all my printers to Klipper I don't want to look back to Marlin.