SkyNet3D/anet-board #10: Linux instructions

kauedg, 2017-09-12:
Please add instructions on how to install this on Linux.

The "anet" folder should go in
`/home/username/Arduino/hardware`

flamablecharlie, 2018-02-19:
I just tried this, and it doesn't work, sadly.

jcress, 2018-04-22:
i've copied the anet directory to ~/.arduino/hardware , ~/Arduino/hardware an /usr/share/arduino/hardware  and the anet option never shows up under tools -> boards

Clemtaur, 2018-11-04:
Running Arduino 1.8.6  under Linux Mint 'Tara' I get:

Could not find boards.txt in /home/clemtaur/arduino-1.8.6/hardware/anet-board-master/hardware. Is it pre-1.5?
WARNING: Error loading hardware folder /home/clemtaur/arduino-1.8.6/hardware/anet-board-master
  No valid hardware definitions found in folder anet-board-master.
???

closed 2021-03-25 by kauedg