SkyNet3D/anet-board #11: Its not working for me

Jordancjb, 2017-11-23:
I don't know what's happening but when I try it says "Error uploading to anet board v.1" or "Error uploading to anet board v.1 (Optiboot)" please help me

jasonshrout, 2018-02-03:
Jordan, 
It looks like it's been awhile and you've probably figured it out already.  If not can you give more detail?

Chris3485, 2018-07-25:
I am trying to add the SkyNet to my Anet A8 and when i get to the burning the bootloader it just comes up saying Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "Anet V1.0 (Optiboot)"

Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

krekr, 2021-07-02:
Same for me (Arduino on Mac); Error while burning bootloader "0xff != 0x0f".
**However**, afterwards, I was able to upload code through Arduino, and everything seems to work. So it seems the bootloader burns alright, it just doesn't pass a check.

closed 2023-02-17 by Jordancjb