SkyNet3D/anet-board #16: Anet A8 V1.5 board Denied

poorbullfrog, 2018-05-16:
I have been trying to update my A8 firmware to marlin. I can see the board in my device manager without errors. How ever I keep getting this error message. Please help.
avrdude: ser_open(): can't open device "\\.\COM4": Access is denied.

DoctorPunto, 2018-05-19:
I am having the same issue. this file is for the V1.0 versions of the board. Do we know if there is a v1-5 in the works?
avrdude: ser_open(): can't open device "\\.\COM5": Access is denied.

poorbullfrog, 2018-05-19:
I double checked the device manager, the board is the only thing requesting a com port. Even when trying to burn a boot loader the request timed out. The com light on the anet board will blink intermittently then cancel out.

poorbullfrog, 2018-05-19:
Doctor punto: I was able to get past the access denied part by. Opening device manager clicking on view, select hidden/disconnected devices. Then right clicking on each listed under com port then removed and deleted drivers. I then used drivers from the sd card.