SkyNet3D/anet-board #20: Compiler warnings: Redefined TIMERXX on Arduino 1.8.8

carlosgalvezp, 2019-03-05:
Hi,

I am trying to build Marlin 1.1.9 using the latest Anet board from this repository, on Arduino 1.8.8.
It seems like Arduino has changed something, because now when building Marlin I get dangerous compilation warnings about redefinition of pins for TIMERs. Here's a short listing of the warnings, but there are more:

```
In file included from /opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from /opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Stream.cpp:25:
/opt/arduino-1.8.8/hardware/anet/avr/variants/sanguino/pins_arduino.h:53:0: warning: "TIMER5A" redefined
 #define TIMER5A 14
 ^
In file included from /opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Stream.cpp:25:0:
/opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Arduino.h:221:0: note: this is the location of the previous definition
 #define TIMER5A 16
 ^
In file included from /opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from /opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Stream.cpp:25:
/opt/arduino-1.8.8/hardware/anet/avr/variants/sanguino/pins_arduino.h:54:0: warning: "TIMER5B" redefined
 #define TIMER5B 15
 ^
In file included from /opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Stream.cpp:25:0:
/opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Arduino.h:222:0: note: this is the location of the previous definition
 #define TIMER5B 17
 ^
In file included from /opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from /opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Stream.cpp:25:
/opt/arduino-1.8.8/hardware/anet/avr/variants/sanguino/pins_arduino.h:55:0: warning: "TIMER5C" redefined
 #define TIMER5C 16
 ^
In file included from /opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Stream.cpp:25:0:
/opt/arduino-1.8.8/hardware/arduino/avr/cores/arduino/Arduino.h:223:0: note: this is the location of the previous definition
 #define TIMER5C 18
```

You can see for example that Arduino defines TIMER5C to 18 and SkyNet redefines it to 16. Which one is correct? Can this be fixed here?

Thanks!

bettercomb, 2019-04-09:
It would appear the renumbered TIMERXX values in Arduino.h of the ArduinoCore-avr tree occurred back in 2011 via commit 5bbe2a63e48ec3057322167a3a6745a66f0c8941 which introduced a new TIMER4D value, and in 2014 via commit 3a48c9240d33a32f7afdcb06ac345fa24d4aa46b which introduced a new TIMER1C value, incrementing the following TIMERXX values by 1 in both cases.
 
- Do these expansions affect the processing supported on the Anet board? 
- Does it affect different versions of the board differently (e.g., V1.0 original, V1.5 current), such that it needs to be reflected with appropriate #ifdef/#endif bracketing in pins_arduino.h?

A simple "#if 0/#endif" around the duplicate definitions in sanguino/pins_arduino.h eliminates the warnings, but I'm not sufficiently knowledgeable about the board-specific processing to know the full effects of such a change. 

It should also be noted that the standard/pins_arduino.h lacks these same TIMERXX redefinitions of sanguino/pins_arduino.h.

benlye, 2019-04-11:
I think it's safe to remove the TIMERXX definitions from the Anet board's pins file.  I've just tested three different compilations: 
* PlatformIO
* Arduino with the board TIMERXX defintions
* Arduino with the TIMERXX definitions removed  from the board's pins file (so using the default TIMERXX definitions from the Arduino AVR core)

I compiled with the standard Anet A8 example config with `PINS_DEBUGGING` enabled and used `M43 I` to check the pin assignments.  In all three cases it was exactly the same:

```
PIN:   0   Port: B0        E0_DIR_PIN                             Output = 1  
PIN:   1   Port: B1        E0_STEP_PIN                            Output = 0  
PIN:   2   Port: B2        Z_DIR_PIN                              Output = 1  
PIN:   3   Port: B3        Z_STEP_PIN                             Output = 0    TIMER0A   PWM:     0    WGM: 3    COM0A: 0    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled  
PIN:   4   Port: B4        AVR_SS_PIN                             Output = 0    TIMER0B   PWM:   128    WGM: 3    COM0B: 0    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow interrupt enabled  
.                          FAN_PIN                                Output = 0
.                          SS_PIN                                 Output = 0
PIN:   5   Port: B5        AVR_MOSI_PIN                           Input  = 0  
.                          MOSI_PIN                               Input  = 0
PIN:   6   Port: B6        AVR_MISO_PIN                           Input  = 1  
.                          MISO_PIN                               Input  = 1
PIN:   7   Port: B7        AVR_SCK_PIN                            Input  = 0  
.                          SCK_PIN                                Input  = 0
PIN:   8   Port: D0        RXD                                    Input  = 1  
PIN:   9   Port: D1        TXD                                    Input  = 0  
PIN:  10   Port: D2        LCD_PINS_D4                            Output = 0  
PIN:  11   Port: D3        LCD_PINS_D5                            Output = 0  
PIN:  12   Port: D4        HEATER_BED_PIN                         Output = 0    TIMER1B   PWM:     0    WGM: 4    COM1B: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode  
PIN:  13   Port: D5        HEATER_0_PIN                           Output = 0    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled  
PIN:  14   Port: D6        E0_ENABLE_PIN                          Output = 1    TIMER2B   PWM:     0    WGM: 1    COM2B: 0    CS: 4    TCCR2A: 1    TCCR2B: 4    TIMSK2: 0  
.                          X_ENABLE_PIN                           Output = 1
.                          Y_ENABLE_PIN                           Output = 1
PIN:  15   Port: D7        X_STEP_PIN                             Output = 0    TIMER2A   PWM:     0    WGM: 1    COM2A: 0    CS: 4    TCCR2A: 1    TCCR2B: 4    TIMSK2: 0  
PIN:  16   Port: C0        LCD_PINS_D6                            Output = 0  
PIN:  17   Port: C1        LCD_PINS_D7                            Output = 0  
PIN:  18   Port: C2        X_MIN_PIN                              Input  = 1  
.                          X_STOP_PIN                             Input  = 1
PIN:  19   Port: C3        Y_MIN_PIN                              Input  = 1  
.                          Y_STOP_PIN                             Input  = 1
PIN:  20   Port: C4        Z_MIN_PIN                              Input  = 1  
.                          Z_STOP_PIN                             Input  = 1
PIN:  21   Port: C5        X_DIR_PIN                              Output = 1  
PIN:  22   Port: C6        Y_STEP_PIN                             Output = 0  
PIN:  23   Port: C7        Y_DIR_PIN                              Output = 1  
PIN:  24   Port: A7 (A 7)  TEMP_0_PIN                             Analog in =  1023  
PIN:  25   Port: A6 (A 6)  TEMP_BED_PIN                           Analog in =  1023  
PIN:  26   Port: A5 (A 5)  Z_ENABLE_PIN                           Output = 1  
PIN:  27   Port: A4 (A 4)  SERVO0_PIN                             Input  = 0  
PIN:  28   Port: A3 (A 3)  LCD_PINS_RS                            Output = 1  
.                          LCD_SDSS                               Output = 1
PIN:  29   Port: A2 (A 2)  LCD_PINS_ENABLE                        Output = 0  
PIN:  30   Port: A1 (A 1)  <unused/unknown>   Analog in =  1020   Input  = 1  
PIN:  31   Port: A0 (A 0)  SDSS                                   Input  = 0  
ok
```
[Anet Pins - PlatformIO.txt](https://github.com/SkyNet3D/anet-board/files/3068186/Anet.Pins.-.PlatformIO.txt)
[Anet Pins - Arduino TIMERs.txt](https://github.com/SkyNet3D/anet-board/files/3068187/Anet.Pins.-.Arduino.TIMERs.txt)
[Anet Pins - board TIMERs.txt](https://github.com/SkyNet3D/anet-board/files/3068188/Anet.Pins.-.board.TIMERs.txt)

I'm going to update my Arduino IDE Boards Manager package (https://github.com/benlye/anet-board/) to remove the TIMERXX definitions.