SkyNet3D/anet-board #22: cant compile with latest boards from arduino

bj97301, 2019-09-18:
please see screenshots below
[image]
[image]

benlye, 2019-09-20:
The board here is no longer maintained.

Use this one instead, which does not have this issue.
https://github.com/benlye/anet-board

CpuID, 2020-07-02:
For anyone else who ends up here: https://github.com/benlye/anet-board currently doesn't fix this issue. Same compile issue, tested with Marlin 1.1.9.1 1314b31d97bba8cd74c6625c47176d4692f57790

benlye, 2020-07-02:
@CpuID Do you really get the same error as posted above?  I just tried to compile the same tag.  It failed but I got totally different errors to the TIMERx redefines mentioned above (which are fixed).

CpuID, 2020-07-02:
@benlye I apologise, your repo does actually change the error profile. Currently what I get:

```
In file included from sketch/endstops.cpp:35:0:
sketch/endstop_interrupts.h: In function 'void setup_endstop_interrupts()':
endstop_interrupts.h:121:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(X_MIN_PIN) != NULL, "X_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:258:0,
                 from sketch/HAL.h:36,
                 from sketch/MarlinConfig.h:39,
                 from sketch/Marlin.h:35,
                 from sketch/endstops.cpp:27:
/Users/nathan/Library/Arduino15/packages/anet/hardware/avr/1.0.2/variants/sanguino/pins_arduino.h:92:66: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)    ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                 ~^~~~~~~
sketch/endstop_interrupts.h:121:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(X_MIN_PIN) != NULL, "X_MIN_PIN is not interrupt-capable");
                     ^~~~~~~~~~~~~~~~~
In file included from sketch/endstops.cpp:35:0:
endstop_interrupts.h:141:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(Y_MIN_PIN) != NULL, "Y_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:258:0,
                 from sketch/HAL.h:36,
                 from sketch/MarlinConfig.h:39,
                 from sketch/Marlin.h:35,
                 from sketch/endstops.cpp:27:
/Users/nathan/Library/Arduino15/packages/anet/hardware/avr/1.0.2/variants/sanguino/pins_arduino.h:92:66: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)    ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                 ~^~~~~~~
sketch/endstop_interrupts.h:141:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(Y_MIN_PIN) != NULL, "Y_MIN_PIN is not interrupt-capable");
                     ^~~~~~~~~~~~~~~~~
In file included from sketch/endstops.cpp:35:0:
endstop_interrupts.h:161:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(Z_MIN_PIN) != NULL, "Z_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:258:0,
                 from sketch/HAL.h:36,
                 from sketch/MarlinConfig.h:39,
                 from sketch/Marlin.h:35,
                 from sketch/endstops.cpp:27:
/Users/nathan/Library/Arduino15/packages/anet/hardware/avr/1.0.2/variants/sanguino/pins_arduino.h:92:66: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)    ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                 ~^~~~~~~
sketch/endstop_interrupts.h:161:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(Z_MIN_PIN) != NULL, "Z_MIN_PIN is not interrupt-capable");
                     ^~~~~~~~~~~~~~~~~
exit status 1
non-constant condition for static assertion
```

I then tried applying this patch (partial of https://github.com/MarlinFirmware/Marlin/pull/15771 ):

```

```

And ended up with the same error (I think?):

```
In file included from sketch/endstops.cpp:35:0:
sketch/endstop_interrupts.h: In function 'void setup_endstop_interrupts()':
endstop_interrupts.h:126:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(X_MIN_PIN) != NULL, "X_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:258:0,
                 from sketch/HAL.h:36,
                 from sketch/MarlinConfig.h:39,
[... 18 lines cut ...]
 #define digitalPinToPCICR(p)    ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                 ~^~~~~~~
sketch/endstop_interrupts.h:146:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(Y_MIN_PIN) != NULL, "Y_MIN_PIN is not interrupt-capable");
                     ^~~~~~~~~~~~~~~~~
In file included from sketch/endstops.cpp:35:0:
endstop_interrupts.h:166:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(Z_MIN_PIN) != NULL, "Z_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:258:0,
                 from sketch/HAL.h:36,
                 from sketch/MarlinConfig.h:39,
                 from sketch/Marlin.h:35,
                 from sketch/endstops.cpp:27:
/Users/nathan/Library/Arduino15/packages/anet/hardware/avr/1.0.2/variants/sanguino/pins_arduino.h:92:66: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)    ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                 ~^~~~~~~
sketch/endstop_interrupts.h:166:21: note: in expansion of macro 'digitalPinToPCICR'
[... cut ...]
                     ^~~~~~~~~~~~~~~~~
exit status 1
non-constant condition for static assertion
```

I was considering if I need to make the leap to 2.0, was trying to put it off (for a trivial `Configuration.h` change), but I might need to bite the bullet... :( I'll read over these errors a little closer first and see if there's a simple answer. https://github.com/MarlinFirmware/Marlin/issues/15770 also seems to have a bunch of comments in there and suggestions etc I haven't completely exhausted.

CpuID, 2020-07-02:
My next test, was trying to drop the version of Arduino AVR Boards in Board Manager from 1.8.3 -> 1.6.23. But then I hit this (I believe due to MacOS Catalina no longer running 32bit binaries, FML):

```
fork/exec /Users/nathan/Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-g++: bad CPU type in executable
Error compiling for board Anet V1.0.
```

CpuID, 2020-07-02:
Original error in https://github.com/SkyNet3D/anet-board/issues/22#issuecomment-652976024 when using Arduino AVR Boards 1.8.2 and 1.8.1.

Idea: Copy the `~/Library/Arduino15/packages/arduino/tools/avr-gcc` from Arduino AVR Boards 1.8.1 over to 1.6.23 (doesn't seem included with 1.8.2 or 1.8.3?). Leads to the same error/s as Arduino AVR Boards 1.8.x.

CpuID, 2020-07-02:
Short of other suggestions here, I think I'm going to spend my time getting Marlin 2.0 installed instead. Probably better to just embrace the future rather than hanging onto the past with this one... (more active development etc)